TOSUN-Shanghai/TSMaster #1: 建议增加一个feature：允许“graphics/trace/...”作为一个独立的窗口

ilikerome, 2021-05-21:
一般都要同时开多个窗口，一个屏幕太挤了。

freshhope, 2021-05-23:
你的意思是将graphics和trace合并？可以这么做：下方新建一页，在新建页面中只添加graphics和trace，然后工程-水平排列或垂直排列即可

ilikerome, 2021-05-26:
我的意思是，将grahics / trace作为一个独立的windows窗口，每个窗口都可以独立的移动到某个屏幕上。
因为实际回放log既要看着trace，也要看着某个signal graphics的变化情况，还要去点击其他的窗口按钮之类的操作，一个电脑屏幕并排显示多个窗口太小了，实际使用 CxxOE的过程中，需要频繁的拖放各个组件窗口。

1、另外一个解决方案是使用27寸及以上的显示器，扩大显示面积  2、多窗口可以独立在显示器上放大缩小放置

freshhope, 2021-05-29:
理解，这个提议不错，我考虑下实现

freshhope, 2021-05-31:
试一下今天发布的beta版

http://www.tosun.tech/TOSUNSoftware/TSMaster_Setup_beta.exe

一个布局窗口内部可以增加无限种子窗口组合，这样的布局窗口也可以创建无限个：
[image]

VS2005的布局风格：
[image]

freshhope, 2021-06-18:
好的，我们努力改进。

closed 2021-06-18 by freshhope

ilikerome, 2021-06-18:
> 好的，我们努力改进。

ok