TOSUN-Shanghai/TSMaster #2: WinUSB WCID support

kubsztal, 2021-06-02:
Would be rally great if besides supported commercial CAN vendors you could support WinUSB WCID protocol, which is used by several open source CAN projects, e.g.:
https://github.com/candle-usb/candleLight_fw
https://canable.io/
https://github.com/HubertD/cangaroo

freshhope, 2021-06-04:
Thank you very much for the suggestion! please wait until this feature is out.

freshhope, 2021-10-06:
Please see release on 2021-10-06

[image]

[image]

Rx Benchmark under 500 Kbps:
CH1: CANable
CH2: TC1005
TC1005 tx around 10250 frames per second (100% bus load utilized, TC1005 performance > 100000 frames per second in all its 5 CAN channels)
CANable rx around 10250 frames per second (single rx performance is OK)
[image]

Tx Benchmark under 500 Kbps:
CANable tx around 2700 frames per second (max tx frame rate should be 10250)
[image]

closed 2021-10-10 by freshhope