TOSUN-Shanghai/TSMaster #5: GCAN Support

kobzar, 2021-09-27:
Please, add support for GCAN devices like USBCAN-II C 
(https://aliexpress.ru/item/4001216169363.html?spm=a2g0o.store_pc_home.hotSpots_6000242637595.0)

freshhope, 2021-09-27:
I know this kind of device coming from Guangcheng Tech. I'll add support if i get one in the future.

alex-arc-boats, 2021-11-09:
@kobzar [this](https://www.aliexpress.com/item/1005001909644558.html?spm=a2g0o.productlist.0.0.6a236e0eXeMeYZ&algo_pvid=01857279-7963-4fc6-9fa2-7d04128a2e8d&algo_expid=01857279-7963-4fc6-9fa2-7d04128a2e8d-39&btsid=0bb0622e16190325787882656e7766&ws_ab_test=searchweb0_0,searchweb201602_,searchweb201603_) is the same as PCAN and works already

pesky-cpu, 2021-11-09:
@kobzar [this](https://www.aliexpress.com/item/1005001909644558.html?spm=a2g0o.productlist.0.0.6a236e0eXeMeYZ&algo_pvid=01857279-7963-4fc6-9fa2-7d04128a2e8d&algo_expid=01857279-7963-4fc6-9fa2-7d04128a2e8d-39&btsid=0bb0622e16190325787882656e7766&ws_ab_test=searchweb0_0,searchweb201602_,searchweb201603_) is the same as PCAN and works with TSmaster already

closed 2021-11-10 by freshhope