TOSUN-Shanghai/TSMaster #12: J1939 Multiplex Signal Handling

pesky-cpu, 2021-11-09:
**Issue:** For multiplexed signals currently, you have to add the signal to a ‘Numeric Display’, which takes a long time to set up if trying to see multiple multiplexed messages:
**Bug fix:** The ‘Numeric Display’ only updates the bottom value if it’s a multiplexed message. On the below screenshot, the ‘a)’ ‘Battery_voltage’, is updated correctly. But the ‘b)’ ‘Feedback_Current_2’ and ‘Diagnostic_2’ are not updated. I have included the log, DBC, and .t7z configuration for diagnosis, along with a video showing clearly this issue.
**Feature Request:** Have a toggleable option somewhere on the ‘CAN / CAN FD Trace’ window to have persistent memory for multiplexed messages, vs displaying ‘N/A’ when data isn’t received. It might be worthwhile having a toggle to show absolute time received for each signal?
[image]

freshhope, 2022-01-12:
the multiplex engine is fully re-written, so could you please upg within software to latest beta version and check if this issue still exists?

pesky-cpu, 2022-02-10:
Way better - thanks! Only issue is that if replaying a log and you have a message collapsed, if you open the message it will only show the most recent signal. Is it possible to fix this?
[image]

freshhope, 2022-02-13:
Have you try the chronological mode? You need to use this mode to view message history.
[image]

The message list max length may have limitation, if you want to view more message, you can create a new trace and let it only show certain id.

I am also trying to implement x64 version and unlimited message buffer, please wait for the new features.

closed 2022-02-13 by freshhope