TOSUN-Shanghai/TSMaster #17: Transmitting J1939 Multiplexed message

pesky-cpu, 2021-11-09:
**Feature Request:** On the ‘CAN / CAN FD Transmit’ window, if a message is a multiplexed message, only display the signals for the specified multiplex value. Currently all the signals are shown which can be confusing, suggest adding a toggle which shows ‘All’ or ‘Current’ signals, based upon the current multiplexor value.

[image]
[image]

freshhope, 2022-01-12:
the multiplex engine is fully re-written, so could you please upg within software to latest beta version and check if this issue still exists?

ColinGong123, 2024-07-05:
we recommand you try to update the latest version V2024.6.27.1138 and the issue has been fixed.