TOSUN-Shanghai/TSMaster #30: Signal generator automatically starts

pesky-cpu, 2021-12-10:
**Bug:** The message signal generator automatically starts when going online on the bus. This can be dangerous if the signal is to command a motor rpm, etc.
[image]

freshhope, 2021-12-11:
The signal generator will never automatically transmit any message until "auto start" checkbox is checked:
[image]

and the following signal generators will only be active after their parent message is being transmitted:
[image]

If you do not want to auto send messages on application connected, please uncheck the auto start icon on top right area to disable auto start.

The default auto send behavior of newly created transmit window is disabled.

pesky-cpu, 2021-12-13:
The above configuration was automatically transmitting with the 'auto start' unchecked. Can you confirm that it looks correct for not transmitting?

[image]

freshhope, 2021-12-13:
It looks correct, I can confirm the automatically transmitted message is not from this transmit window, it may be from RBS (remaining bus simulation), or mini program (configured for auto start), or online replay module (configure for auto start), and many other potential components, could you please check the following information to see which auto start modules are currently being used:

[image]

pesky-cpu, 2021-12-13:
Not seeing any autostart modules on this configuration

[image]

freshhope, 2021-12-14:
Yes, so could you take a screenshot of both CAN transmit and CAN trace to see which messages are marked as transmit ?

freshhope, 2022-01-17:
closed due to no evidence, reopen if reproduction is possible.

closed 2022-01-17 by freshhope