TOSUN-Shanghai/TSMaster #42: Add support for UDS norm (diagnostic services)

dragoudin, 2022-04-28:
Having ability to upport UDS diagnostic services (frame identification from DBC).
Having an interface to send request/receive response

XieYY135, 2022-04-30:
TSMaster Diagnostic Console module provide graphical interface for user to execute diagnostic services：
[image]

XieYY135, 2022-04-30:
If necessary, the tsmaster also has built-in API function interfaces for diagnostic operations
[image]

totocaca123, 2022-04-30:
I confirm I can create my diagnostics database (working fine) in basic diagnostic config.