TOSUN-Shanghai/TSMaster #48: Diagnostic ISO TP application - service defined not visible in Diagnostic console tab

totocaca123, 2022-04-29:
I added multiple service in tab "Basic Diagnostic Config".
I want to run these services in Diagnostic console, but they are not visible (list is empty)
[image]

Moreover, it would be good to have something to send periodically tester present (a checkbox) to remain in same session during manual testing

XieYY135, 2022-04-30:
[image]

totocaca123, 2022-04-30:
Ok I confirm it's working (the icon validate is not intuitive - an icon REFRESH or synchronize like this would more explicit)
[image]