TOSUN-Shanghai/TSMaster #51: Bug when converting arxml to DBC (Automotive converter)

dragoudin, 2022-05-04:
I have a correct system extract (*.arxml from systemDesk) and I try to convert it into dbc.

An error occurs (message displayed).
In my arxml file, can message ID are in hex format (0x) and it cause an error
extract of my arxml file!
```
                        <CAN-FRAME-TRIGGERING T="xxx" UUID="xxx">
                          <SHORT-NAME>xxxxx</SHORT-NAME>
                          <FRAME-PORT-REFS>
                            <FRAME-PORT-REF DEST="FRAME-PORT">/EcuInstances/ECU/CanCommunicationConnector/RxFramePort</FRAME-PORT-REF>
                          </FRAME-PORT-REFS>
                          <FRAME-REF DEST="CAN-FRAME">/Frames/G_CAN/PKG_G_CAN/Can_Frame_Diag_Functional_Request</FRAME-REF>
                          <PDU-TRIGGERINGS>
                            <PDU-TRIGGERING-REF-CONDITIONAL>
                              <PDU-TRIGGERING-REF DEST="PDU-TRIGGERING">/CommunicationClusters/G_CAN/G_CAN/G_CAN/PduTriggering_DIAG_RQ_GLOBAL_UDS</PDU-TRIGGERING-REF>
                            </PDU-TRIGGERING-REF-CONDITIONAL>
                          </PDU-TRIGGERINGS>
                          <CAN-ADDRESSING-MODE>STANDARD</CAN-ADDRESSING-MODE>
                          **<IDENTIFIER>0x777</IDENTIFIER>**
                        </CAN-FRAME-TRIGGERING>
```

I capture a log message which indicate a problem (tools expect an int data, and in file, there is a hex data)
Source file selected: XXX_SystemExtract.arxml
```
2022.4.28.741 INFO - convert - Importing C:\Users\XXX\AppData\Local\Temp\~LC346D.arxml ... 
2022.4.28.741 Traceback (most recent call last):
2022.4.28.741   File "C:\Program Files (x86)\TOSUN\TSMaster\bin\Data\Python\3.8.5\x86\lib\runpy.py", line 194, in _run_module_as_main
2022.4.28.741     return _run_code(code, main_globals, None,
2022.4.28.741   File "C:\Program Files (x86)\TOSUN\TSMaster\bin\Data\Python\3.8.5\x86\lib\runpy.py", line 87, in _run_code
2022.4.28.741     exec(code, run_globals)
2022.4.28.741   File "C:\Program Files (x86)\TOSUN\TSMaster\bin\Data\Python\3.8.5\x86\Scripts\canconvert.exe\__main__.py", line 7, in <module>
2022.4.28.741   File "C:\Program Files (x86)\TOSUN\TSMaster\bin\Data\Python\3.8.5\x86\lib\site-packages\click\core.py", line 829, in __call__
2022.4.28.741     return self.main(*args, **kwargs)
2022.4.28.741   File "C:\Program Files (x86)\TOSUN\TSMaster\bin\Data\Python\3.8.5\x86\lib\site-packages\click\core.py", line 782, in main
2022.4.28.741     rv = self.invoke(ctx)
2022.4.28.741   File "C:\Program Files (x86)\TOSUN\TSMaster\bin\Data\Python\3.8.5\x86\lib\site-packages\click\core.py", line 1066, in invoke
2022.4.28.741     return ctx.invoke(self.callback, **ctx.params)
2022.4.28.741   File "C:\Program Files (x86)\TOSUN\TSMaster\bin\Data\Python\3.8.5\x86\lib\site-packages\click\core.py", line 610, in invoke
2022.4.28.741     return callback(*args, **kwargs)
2022.4.28.741   File "C:\Program Files (x86)\TOSUN\TSMaster\bin\Data\Python\3.8.5\x86\lib\site-packages\canmatrix\cli\convert.py", line 135, in cli_convert
2022.4.28.741     canmatrix.convert.convert(infile, outfile, **options)
2022.4.28.741   File "C:\Program Files (x86)\TOSUN\TSMaster\bin\Data\Python\3.8.5\x86\lib\site-packages\canmatrix\convert.py", line 40, in convert
2022.4.28.741     dbs = canmatrix.formats.loadp(infile, **options)
2022.4.28.741   File "C:\Program Files (x86)\TOSUN\TSMaster\bin\Data\Python\3.8.5\x86\lib\site-packages\canmatrix\formats\__init__.py", line 77, in loadp
2022.4.28.741     return load(fileObject, import_type, key, **options)
2022.4.28.741   File "C:\Program Files (x86)\TOSUN\TSMaster\bin\Data\Python\3.8.5\x86\lib\site-packages\canmatrix\formats\__init__.py", line 94, in load
2022.4.28.741     dbs = module_instance.load(file_object, **options)  # type: ignore
2022.4.28.741   File "C:\Program Files (x86)\TOSUN\TSMaster\bin\Data\Python\3.8.5\x86\lib\site-packages\canmatrix\formats\arxml.py", line 1963, in load
2022.4.28.741     result.update(decode_can_helper(ea, float_factory, ignore_cluster_info))
2022.4.28.741   File "C:\Program Files (x86)\TOSUN\TSMaster\bin\Data\Python\3.8.5\x86\lib\site-packages\canmatrix\formats\arxml.py", line 1858, in decode_can_helper
2022.4.28.741     frame = get_frame(frameTrig, ea, multiplex_translation, float_factory)
2022.4.28.741   File "C:\Program Files (x86)\TOSUN\TSMaster\bin\Data\Python\3.8.5\x86\lib\site-packages\canmatrix\formats\arxml.py", line 1447, in get_frame
2022.4.28.741     arbitration_id = int(arb_id.text)
2022.4.28.741 ValueError: invalid literal for int() with base 10: '0x777'
2022.4.28.741 Error: return code = 1
```

totocaca123, 2022-05-04:
This is a canmatrix issue

It is already corrected here https://github.com/ebroecker/canmatrix/pull/639

So the solution would be to upgrade the version of canmatrix package used by TSMaster

freshhope, 2022-05-04:
Thanks! I will upgrade soon

jayjjsun, 2022-11-22:
Hi all

We face same issue when converting arxml to dbc. with TSMaster V2022.6.15
源文件已選擇: simple_demo_can.arxml
2022.6.15.763 INFO - convert - Importing C:\Users\zhija\AppData\Local\Temp\~LCD26F.arxml ... 
2022.6.15.763 Traceback (most recent call last):
2022.6.15.763   File "C:\Program Files (x86)\TOSUN\TSMaster\bin\Data\Python\3.8.5\x86\lib\runpy.py", line 194, in _run_module_as_main
2022.6.15.763     return _run_code(code, main_globals, None,
2022.6.15.763   File "C:\Program Files (x86)\TOSUN\TSMaster\bin\Data\Python\3.8.5\x86\lib\runpy.py", line 87, in _run_code
2022.6.15.763     exec(code, run_globals)
2022.6.15.763   File "C:\Program Files (x86)\TOSUN\TSMaster\bin\Data\Python\3.8.5\x86\Scripts\canconvert.exe\__main__.py", line 7, in <module>
2022.6.15.763   File "C:\Program Files (x86)\TOSUN\TSMaster\bin\Data\Python\3.8.5\x86\lib\site-packages\click\core.py", line 829, in __call__
2022.6.15.763     return self.main(*args, **kwargs)
2022.6.15.763   File "C:\Program Files (x86)\TOSUN\TSMaster\bin\Data\Python\3.8.5\x86\lib\site-packages\click\core.py", line 782, in main
2022.6.15.763     rv = self.invoke(ctx)
2022.6.15.763   File "C:\Program Files (x86)\TOSUN\TSMaster\bin\Data\Python\3.8.5\x86\lib\site-packages\click\core.py", line 1066, in invoke
2022.6.15.763     return ctx.invoke(self.callback, **ctx.params)
2022.6.15.763   File "C:\Program Files (x86)\TOSUN\TSMaster\bin\Data\Python\3.8.5\x86\lib\site-packages\click\core.py", line 610, in invoke
2022.6.15.763     return callback(*args, **kwargs)
2022.6.15.763   File "C:\Program Files (x86)\TOSUN\TSMaster\bin\Data\Python\3.8.5\x86\lib\site-packages\canmatrix\cli\convert.py", line 135, in cli_convert
2022.6.15.763     canmatrix.convert.convert(infile, outfile, **options)
2022.6.15.763   File "C:\Program Files (x86)\TOSUN\TSMaster\bin\Data\Python\3.8.5\x86\lib\site-packages\canmatrix\convert.py", line 40, in convert
2022.6.15.763     dbs = canmatrix.formats.loadp(infile, **options)
2022.6.15.763   File "C:\Program Files (x86)\TOSUN\TSMaster\bin\Data\Python\3.8.5\x86\lib\site-packages\canmatrix\formats\__init__.py", line 77, in loadp
2022.6.15.763     return load(fileObject, import_type, key, **options)
2022.6.15.763   File "C:\Program Files (x86)\TOSUN\TSMaster\bin\Data\Python\3.8.5\x86\lib\site-packages\canmatrix\formats\__init__.py", line 94, in load
2022.6.15.763     dbs = module_instance.load(file_object, **options)  # type: ignore
2022.6.15.763   File "C:\Program Files (x86)\TOSUN\TSMaster\bin\Data\Python\3.8.5\x86\lib\site-packages\canmatrix\formats\arxml.py", line 2017, in load
2022.6.15.763     result.update(decode_can_helper(ea, float_factory, ignore_cluster_info))
2022.6.15.763   File "C:\Program Files (x86)\TOSUN\TSMaster\bin\Data\Python\3.8.5\x86\lib\site-packages\canmatrix\formats\arxml.py", line 1899, in decode_can_helper
2022.6.15.763     uuid = fme.get("UUID")
2022.6.15.763 AttributeError: 'NoneType' object has no attribute 'get'
2022.6.15.763 錯誤：返回值 = 1

Thanks!