TOSUN-Shanghai/TSMaster #53: Diagnostic ISO TP -

totocaca123, 2022-05-05:
When I type manually my commands and press key "Enter", my command in not executed. I need to press the button execute with the mouse.
After pressing "Enter" key, the command line need to be cleared to allow me to put the next command

Moreover, command data needs to be displayed as padded "10 01" insted of "1001" in the input text.

[image]

XieYY135, 2022-05-06:
This feature will be supported in recent releases

XieYY135, 2022-05-07:
[image]

closed 2022-06-04 by XieYY135

dragoudin, 2022-08-19:
I tested this feature in version 2022-08-19

- When I type my command and I press enter, diagnostic command is padded : Issue corrected
- But the command is not executed (I have to press button Execute to send my request).

This issue is partially corrected and execution of command is missing (when I put my command and press enter, it's more intuitive and faster than using mouse to press button Execute).
So this issue needs to be reopened