TOSUN-Shanghai/TSMaster #54: Diagnostic ISO TP - Tab "Automatic Diagnostic" is not working

totocaca123, 2022-05-05:
On the left panell, the tree view is freezed. It would be interesting to add group/subgroup...

Having the possibility to copy/paste some test cases...

Having a button Run and more context menu for each panel (right click with mouse).

Having some button to export test result (format junit for example or excel/word/pdf/html report?)

XieYY135, 2022-05-06:
[image]

dragoudin, 2022-05-06:
I use version V2022.4.28.741 (and when I check update, I have latest version.)
See my screenshot (I have not the same view and no button Unlock/Run and left pannel is freezed.
Moreover, is it possible to add a timeout as test step (wait 1s for example in case of reset)

[image]

XieYY135, 2022-05-07:
> I use version V2022.4.28.741
> 
> Seems it will be activated in a future version

The operation of diagnosis depends on some specific hardwares. How do you test it?

totocaca123, 2022-05-07:
I have same issue on my side. I use vector VN1630 DLL

XieYY135, 2022-05-08:
> 

> In the latest release, V2022.4.28.744 it is corrected

The operation of diagnosis module depends on some specific hardwares. How do you test it?

totocaca123, 2022-05-09:
I use vector vn1610.

jianhun9527, 2022-05-10:
> [image]

Hello, I would like to ask how to implement the 22 F1 88 service in the picture, because the lower computer replies with multiple frames, but I cannot make TSMASTER send flow control frames

XieYY135, 2022-06-17:
> > [image]
> 
> Hello, I would like to ask how to implement the 22 F1 88 service in the picture, because the lower computer replies with multiple frames, but I cannot make TSMASTER send flow control frames

Please use the hardware from TOSUN Because the transport layer is implemented in the firmware for better time accuracy.