TOSUN-Shanghai/TSMaster #58: TSMASTER sends flow control frame

jianhun9527, 2022-05-10:
How can I configure the diagnostic panel to automatically send flow control frames under the UDS service

[image]

22 The reason for the service failure is that the flow control frame is not sent to the lower computer

dragoudin, 2022-05-11:
I confirm the issue. When I read a DID (22 XX XX service for example), if the expected response is big (more than 8 byte which requires multiple messages), only the first message is received in trace. TSMaster doesn't send the flow control frame.
I am wondering if the settings of tp can change anything?

XieYY135, 2022-05-19:
In order to ensure timeliness, the transport layer is implemented in firmware. As follows:
[image]

dragoudin, 2022-05-20:
> In order to ensure timeliness, the transport layer is implemented in firmware. As follows: [image]

I use the latest version of TSMaster (2022.5.8.746) and it is not working on my side

XieYY135, 2022-05-20:
> 

This function only supports specific hardware devices

XieYY135, 2022-05-20:
> > In order to ensure timeliness, the transport layer is implemented in firmware. As follows: [image]
> 
> I use the latest version of TSMaster (2022.5.8.746) and it is not working on my side

This function only supports specific hardware devices

totocaca123, 2022-05-21:
I use vector VN1630 hardware and it is not working.

XieYY135, 2022-05-29:
> > > 
> > 
> > 
> > Hardware devices from TOSUN are required.
> 
> If this feature is not working with other hardware than TOSUN, it is an issue (it's not possible to send/receive UDS services bigger than 8 bytes). Send flow is a basic feature and if hardware is not compatible, I believe it can be done by software

In the current design, the diagnostic transport layer is implemented in the hardware.

totocaca123, 2022-05-31:
> In the current design, the diagnostic transport layer is implemented in the hardware.

I understand. It seems than Tosun hardware support natively flow control from hardware. 
But for people using different hardware (peak, vector...) this feature to test CanTp and diagnostic services is quiet important, and a such feature would be very interesting

XieYY135, 2022-06-03:
> I understand. It seems than Tosun hardware support natively flow control from hardware. But for people using different hardware (peak, vector...) this feature to test CanTp and diagnostic services [...]

After our test, we found that the timestamp of some hardware(of course not vector) is not so accurate, which is also the reason why we do not support it

totocaca123, 2022-06-03:
> After our test, we found that the timestamp of some hardware(of course not vector) is not so accurate, which is also the reason why we do not support it

There might be some performance issue for the send flow (it is an hardware limitation). But if TsMaster doesn't support Send flow, it cause au problem for testing (most part of Diagnostic services UDS are not available without a tosun hardware).

totocaca123, 2022-06-07:
A feature would be to enable a software send flow feature (if not supported by hardware)

jianhun9527, 2022-06-12:
The flow control frame can be implemented through the C code editor, as follows, which has been verified
[image]

closed 2022-06-12 by jianhun9527