TOSUN-Shanghai/TSMaster #64: Absolute Timestamp as unix timestamp?

Matheus-Garbelini, 2022-05-19:
Hello, 

We are trying to save frames using absolute unix timestamp instead of relative, which starts from 0.
Is it possible to configure somewhere to absolute timestamp or change this after saving the file log via TSmaster SDK?

XieYY135, 2022-05-20:
You can use the API to read the initial time of the record file, and add the relative time to get the absolute time

Matheus-Garbelini, 2022-05-20:
@XieYY135 Thanks, can you point what is the function to get the initial time of the record file?
Is it specific to BLF format?

Matheus-Garbelini, 2022-05-23:
Hi @XieYY135, any idea if we can use **FTimeUs** to inform our own hardware timestamp?

Matheus-Garbelini, 2022-05-26:
@XieYY135 I'm using TSMaster GUI and miniprogram, not the standalone SDK API.
Yes, if I use the API to write a custom CAN Frame timestamp to `tslog_blf_write_canfd`, than it works.
This means that I cannot use the "Bus Logger" from TSMaster GUI. Is there atleast a system variable or miniprogram that can inform us what is the log file path the user specified in the "Bus Logger"?
This way the user doesn't need to manually modify the miniprogram code to change log file path.
[image]

freshhope, 2022-05-26:
This system var will be available in the next release

Matheus-Garbelini, 2022-05-26:
Thanks a lot @freshhope . Will close this issue. Will get back on my previous issue with more information which I think can be solved in the next release.

closed 2022-05-26 by Matheus-Garbelini

Matheus-Garbelini, 2022-06-17:
Hi @freshhope, @XieYY135 
I'm trying to contact TSMaster to check for the possibility of a collaboration with our university for a CAN hardware that we are developing. Is there any email that I can reach for collaboration inquiries?