TOSUN-Shanghai/TSMaster #83: Support for multiplex ranges

Zultanic, 2022-11-10:
Hi, Thank you for the awesome software. 

**Request:** I am trying to use multiplexing where the multiplexed signal can have a range of values that activate it.
This is supported in dbc files like with the entry shown below.
SG_MUL_VAL_ 2566573056 multiplexed_signal multiplexor_signal 0-0, 8-8;

**Current Behavior:** The multiplexed_signal is only activated when the multiplex value in "m0" or "m8" is used and does not consider the other numbers in the range

freshhope, 2022-11-11:
Hi Zultanic,

could you provide this dbc for me to check? Thank you

Zultanic, 2022-11-14:
Hi freshhope, thanks for the response

I have attached an example [dbc](https://github.com/TOSUN-Shanghai/TSMaster/files/10006327/MultiplexExampleDBC.dbc.txt) file that I hope illustrates the issue. I have one message, one multiplexor signal and three multiplexed examples. 
- First example is standard multiplexing, this one works well in TSMaster.
- Second example is trying to use ranges of values defined at the bottom of file by SG_MUL_VAL_
- Third example is similar to ranges but has multiple non-continuous multiplex numbers once again defined by SG_MUL_VAL_

I also found [this](https://cdn.vector.com/cms/content/know-how/_application-notes/AN-ION-1-0521_Extended_Signal_Multiplexing.pdf) documentation on extended multiplexing from Vector, maybe it will help to understand the formatting.

ColinGong123, 2024-07-05:
Hi Zultanic,We tested using the latest version of TSMaster V2024.6.27, and the trace window was able to properly parse all values of the multiplexer. If you have any questions during testing, you can leave your email and we will provide assistance through remote control.
[image]