TOSUN-Shanghai/TSMaster #84: [Bug][Diagnostic][Diagnostic Console] Control Flow

dragoudin, 2022-11-10:
I use hw vector VN1630.
Control Flow is not supported by hardware.
I implemented a software program flow (as recomended in issue #58 )
[image]

When I look at the trace, it works fine:
[image]

But in Diagnostic Console, it's still not working.
[image]
I don't view the response (the response of 23 bytes is OK in the raw trace

freshhope, 2022-11-11:
@XieYY135

totocaca123, 2022-12-26:
any news on this topic?

ColinGong123, 2024-07-05:
For other brands of hardware to implement diagnostic flow control frames, it is necessary to purchase a STANDARD subscription version. For specific models, please refer to the link: https://license.tsmaster.net/mainInterface/pricing