TOSUN-Shanghai/TSMaster #85: [FeatureRequest][C Code Editor]??? Call script from panel

dragoudin, 2022-11-10:
I would like to monitor a power supply (RS232). I have a script python to manage this.
I would like to call a python script from panel (a push button + a function callback).
Is there an existing API for this?
When I use the python addon, it seems it is to use a python script to minitor TsMaster.

freshhope, 2022-11-10:
you can use C API, please search "python" to filter out 2 python APIs, steps:
[1] in panel, associate a button with a system variable
[2] in c code editor, implement on var change callback for this system variable
[3] in this callback, you can call your python scripts, such as:

// note: x64 distribution is not installed by default, you need to manually copy folder to bin\Data\Python\active_version\x64
char* s =
"\
import sys\n\
import win32api, win32con\n\
win32api.MessageBox(0, 'test with arguments: ' + sys.argv[1] + ' ' + sys.argv[2], 'title', win32con.MB_OK)\n\
print('OK')\n\
";

char* p;     
if (app.check(app.execute_python_string(s, "arg1 arg2", true, false, &p))){
  log("execution result is %s", p);
} else {
  log("execution failed with result %s", p);
}

totocaca123, 2022-11-11:
I checked and it works fine. It use python 3.8.5 x86 from TsMaster install (is there a way to change python env used?)

Now, if the script doesn't exist or if there is a script error, TsMaster crash (script file doesn't exist)
```
if (app.check(app.execute_python_script("C:\\xxxxx.py", "arg1 arg2", true, false, &p))){
 log("execution result is %s", p);
} else {
  log("execution failed with result %s", p);
}
```
[image]

[image]

freshhope, 2022-11-12:
_(is there a way to change python env used?)_
Yes, just replace "TSMaster\bin\Data\Python\3.8.5\x86" with your python distribution, as currently "3.8.5" is hard coded in TSMaster, maybe changed in the future. You may also copy some "site-packages" in your new distribution as they are responsible for dbc and arxml database loading.

_TsMaster crash (script file doesn't exist)_
Yes this bug has been fixed and new version will release soon, thank you!