TOSUN-Shanghai/TSMaster #86: [Feature Request] Add hardware support

dragoudin, 2022-11-10:
Currently, the system support multiple CAN/LIN/Flexray hardware.
For testing purposes, we might use power supply (monitored by CAN serial, ethernet).  Maybe some plugins (configurable) to add specific hardware

freshhope, 2022-11-11:
Hi dragoudin,

Yes we support all kinds of standard devices and custom devices, you can build a mini program library, control your custom device in this library, and provide APIs for the entire TSMaster environment.

There is another solution, use TSMaster internal plugin system, however, this plugin interface is not published, maybe available in the future.

totocaca123, 2022-11-11:
OK sounds good.
But how to to create .mp file. How to create a mini program library (is it a compiled C code?). Is it created from C Scripts?

freshhope, 2022-11-12:
steps:
[1] write your code in c code editor and compile
[2] navigate to "properties" sheet and check "Mini Program Library"
[3] then you can view your apis in mini program libray window, and call it in other mini programs, or automation module...**

_Moreover, it would be good to have preinstalled MP lib with common features (manage external power supply, specific board to manage some electrical signal...) It's very specific, so I hardly imagin a standard feature for this_

Yes we have created so many libraries for nearly all kinds of instruments, we want to publish them in the near future