TOSUN-Shanghai/TSMaster #87: [Feature] enum type handling from dbc

totocaca123, 2022-11-11:
I dbc file we have enum types for signals (kind of constants).
It would be good to have access to these values somewhere (importing value tables from dbc into global constants or internal data types)

freshhope, 2022-11-12:
Thanks for suggestion, I will add this feature in system constants

totocaca123, 2022-11-12:
Is it possible to have a better structured constant. For example:

Type1.Value0 (or Type1_Value0)
Type1.Value1

Because we can have multiple enum with value 0, and we dont know which type it is

freshhope, 2023-01-19:
upg. to new version, you can control generation of constant defs.

[image]