TOSUN-Shanghai/TSMaster #88: [Feature][Diagnostic] Add data interpretation for dianostic services

totocaca123, 2022-11-11:
[image]

For data in diag response, it would be good to have fields similar than in ODX file format (offset, gain, unit...)
For value table or formula (for conversion), we cannot convert anything?

Moreover, for some diagnostic services (readDTC), response length might change (1 DTC or multiple). It would be good to have something like sequence (multiple repetition for same type)

freshhope, 2022-11-12:
@XieYY135

ColinGong123, 2024-07-05:
The Diagnosti
[image]
cDB module already supports this feature

ColinGong123, 2024-07-05:
The DiagnosticDB module already supports this feature 
[image]