TOSUN-Shanghai/TSMaster #90: [Bug][CAN FD Playback] CAN FD playback issue when using extended CAN ID

big-jar, 2022-11-24:
Hi,

I met a problem recently when I try to playback the .blf record with extended CAN ID:

when using offline mode, the CAN ID is correct:
[image]

but after switching to online mode for the same .blf file, it seems only support the standard CAN ID:
[image]

Please check the .blf file in attachment. Thanks!

[bug_report_20221124.zip](https://github.com/TOSUN-Shanghai/TSMaster/files/10081455/bug_report_20221124.zip)

freshhope, 2022-11-24:
Hi,

this log file is invalid, because the ids inside are > 0x7FF without extended id format, if you drag this file into CANoe, you will see that the measurement is halted due to invalid id:
[image]

you may notice if extended id is received, a suffix "x" is added to identifier in both TSMaster and CANoe
[image]

big-jar, 2022-11-28:
Hi,

I reproduce the issue and seems find some clues, it may be caused by PeakCAN:

I connect the same DUT and observe the CAN bus in TSMaster, both CANoe VN1630 and PeakCANUSB pro are connected to TSMaster.

you can see that via CANoe the messages are all with 'x' suffix:
[image]

but after I switch to PeakCAN pro, the 'x' disappeared:
[image]

The attachment is the log from these 2 tests. Log recorded by CANoe can be replayed correctly but the log by PeakCAN has the same replay issue. Both log are recorded by TSMaster.

Can you help to check whether it is an compatibility issue between PeakCAN and TSMaster?  Thank you very much!

[logs.zip](https://github.com/TOSUN-Shanghai/TSMaster/files/10100213/logs.zip)

freshhope, 2022-11-28:
Thank you for pointing out this bug in tsmaster peak module, I have corrected this bug, you can test new software in the next release

big-jar, 2022-12-12:
Hi,

I have upgrade to v.2022.12.1.803 and the log function is OK!

but another issue appeared:

I logged a record with correct 'x' suffix for CAN FD extended frames, but it can not be replayed via my PCAN USB:
[image]

But via the same hardware, the previous incorrect record with no 'x' suffix can be replayed:
[image]

Can you help to check it? The 2 records are attached.

[records.zip](https://github.com/TOSUN-Shanghai/TSMaster/files/10206448/records.zip)

freshhope, 2022-12-31:
We don't have peak hardware, but I tried to solve it, could you please download the latest beta version and check if issue still exists?

big-jar, 2023-01-30:
Hi,

I tried v2023.1.15.815 and it worked well! Thank you very much!