TOSUN-Shanghai/TSMaster #91: Source files and malware detection.

talon75, 2022-12-06:
Hi,
I'd like to know where the source files are supposed to be.

Moreover, my Windows Defender just reported TOSUN\TSMaster\bin\TSFibexExplorer.exe with [Backdoor:Win32/Bladabindi!ml](https://www.microsoft.com/en-us/wdsi/threats/malware-encyclopedia-description?name=Backdoor%3aWin32%2fBladabindi!ml&threatid=2147748148).
Without the source files, the only thing I can do is uninstall the (really great) software.

freshhope, 2022-12-06:
Hi,

we apologize, this exe is built with dot net, we also face same problem, we are solving this problem now, please wait the next release.

freshhope, 2022-12-31:
Please update to latest beta version and check if issue still exists