TOSUN-Shanghai/TSMaster #93: [Diagnostic] hw not supported

P208PUG, 2023-01-02:
pcan, valuecan, vector same failure 

`[17804] 2022.12.30.812 Error: Please use TOSUN CAN FD hardware to access licensed features`

dragoudin, 2023-01-03:
I confirm this issue (it is the same issue than #84 ). See issue #58 which explains that control flow it implemented inside TSun hardware, and other devices doesn't have this hw feature (vector, peak...).
I tried to implement control flow by software (issue 84) but it's still not working.

freshhope, 2023-01-03:
@XieYY135

P208PUG, 2023-01-11:
make no sense make sw opensource when option for other HW(vector,valuecan,pcan,)than Tosun_hw are locked ....

ttimasdf, 2023-03-17:
> make no sense make sw opensource when option for other HW(vector,valuecan,pcan,)than Tosun_hw are locked ....

It's not even open source, just abusing Github for distributing / promoting their freeware.

ColinGong123, 2024-07-05:
For other brands of hardware to implement diagnostic flow control frames, it is necessary to purchase a STANDARD subscription version. For specific models, please refer to the link: https://license.tsmaster.net/mainInterface/pricing

closed 2026-09-20 by TSSkyhits