TOSUN-Shanghai/TSMaster #102: add variable in graph for SIL/HIL (matlab)

arohami, 2023-06-05:
I tried to connect connect Matlab to TSMaster following this video: [Matlab Controller SIL HIL](https://www.youtube.com/watch?app=desktop&v=b-d6w2knkjY&ab_channel=TSMaster). I did every step exactly the same and got the same results except for "adding internal variable" to the graph. in the video double clicking the variable adds that to the graph but when i do that, i get this error: `Add failed due to event unsupported: Model1.Out1". note that i already can see the variable's "value" and "last written" data in the "select system variable" window that is opened to choose the variable!

what does this error mean? where can i look for additional information? does this mean that simulation has some problem or it's just the problem for showing on the graph? does it have anything to do with the Matlab's add-ons and toolboxes (should i install any specific one?)?

this is the messages in "system messages" section.

```
22:24:29.503 Application mappings not found or invalid in configuration file
22:24:29.663 New configuration created
22:24:29.733 Application mappings saved: TSMaster
22:24:29.780 Configuration saved: TSMaster
22:24:39.227 Plugin loaded: rtlUIMatlabController.bpl, V2023.5.29.888
22:24:46.801 Press Ctrl + F9 to compile the source, or Press F9 to run the script
22:24:46.866 Validated: Model1
22:25:09.057 Matlab Connected
22:25:09.062 Matlab Connected
22:25:27.890 Validated: simple.slx
22:25:33.769 Validated: ***\***\TSMaster\cache\
22:25:40.122 Validated: ***\***\TSMaster\codegen\
22:26:00.583 Codegen completed: simple
22:26:03.940 Copied: 6 files
22:26:04.581 Mini program opened: C Script Editor [Model1]
22:27:45.780 Source file saved
22:29:39.983 Building: Model1.mp
22:29:39.983 Run without rebuild: Model1
22:29:39.993 Compile success
22:29:40.033 Simulation is running: C Script Editor [Model1]
22:29:40.058 Script is running...
22:29:43.091 CAN 1 bit rate (Kbps) = 500
22:29:43.091 CAN 2 bit rate (Kbps) = 500
22:29:43.140 Bus Statistics timer started
22:29:43.153 Application connected
22:29:55.689 22:29:55: Internal Var Count = 27, User Var Count = 0
22:29:58.267 [1268] 2023.5.29.888 Error: Add failed due to event unspported: Model1.Out1
22:30:45.702 22:30:45: Internal Var Count = 27, User Var Count = 0
```

I could use some help and clarification if possible :)
Thanks

freshhope, 2023-06-05:
please update to latest beta

arohami, 2023-06-05:
> please update to latest beta

Thanks. that actually helped. although i had to copy bunch of header files (rtw_continuous.h, tmwtypes.h, etc) to resolve the errors (i will check later to see whether i can build the code without those header files without any problem). but that update actually solved some of the issues.

arohami, 2023-06-06:
ok. it seems the update will do it all. previously, some header files couldn't be found!

closed 2023-06-06 by arohami