TOSUN-Shanghai/TSMaster #104: problem with vc++

arohami, 2023-06-25:
i followed the instructions in this [video](https://www.youtube.com/watch?v=ibktUcYFUF4&ab_channel=TSMaster) and saved my project as a visual studio project. But when i try to build it i see lots of errors, mostly related to the function added in the mini program.

1. i created a custom function (it works in mini program) named fullBrake. But it seems the return value is wrong!
```
Severity	Code	Description	Project	File	Line	Suppression State
Error	C4716	'fullBrake': must return a value	Model1	...\visual_studio_prj\Model1\Model1.cpp	130	
```
i changed the return value in the VS and the issues was resolved. But why did it happen in the first place? and should i manually do this for every function?! i don't see anywhere in the mini program that i can specify the type of return value but it already is "s32" in the VS project!
the VS function:
```
s32 fullBrake(void) { __try { // Custom Function: 
} __except (1) { log_nok("CRASH detected"); app.terminate_application(); }}
```

2. I use simulink model in the mini program. it's a simple simulink model. and the program already works without any problem in TSMaster. but in the VS prj, when i try to build it, there are errors regarding "extern variables":
```
Severity	Code	Description	Project	File	Line	Suppression State
Error	LNK2019	unresolved external symbol "void __cdecl TSSIL_initialize(void)" (?TSSIL_initialize@@YAXXZ) referenced in function "void __cdecl on_start_newOn_Start1(void)" (?on_start_newOn_Start1@@YAXXZ)	Model1	...\visual_studio_prj\Model1\Model1.obj	1	
Error	LNK2019	unresolved external symbol "void __cdecl TSSIL_step(void)" (?TSSIL_step@@YAXXZ) referenced in function "void __cdecl step(void)" (?step@@YAXXZ)	Model1	...\visual_studio_prj\Model1\Model1.obj	1	
Error	LNK2019	unresolved external symbol "void __cdecl TSSIL_terminate(void)" (?TSSIL_terminate@@YAXXZ) referenced in function "void __cdecl on_stop_NewOn_Stop1(void)" (?on_stop_NewOn_Stop1@@YAXXZ)	Model1	..\visual_studio_prj\Model1\Model1.obj	1	
Error	LNK2001	unresolved external symbol "struct ExtU_TSSIL_T TSSIL_U" (?TSSIL_U@@3UExtU_TSSIL_T@@A)	Model1	..\visual_studio_prj\Model1\Model1.obj	1	
Error	LNK2001	unresolved external symbol "struct ExtY_TSSIL_T TSSIL_Y" (?TSSIL_Y@@3UExtY_TSSIL_T@@A)	Model1	...\visual_studio_prj\Model1\Model1.obj	1	
Error	LNK1120	5 unresolved externals	Model1	...\speed_controller_prj\bin\Model1.mp	1	
```
but the header files and .c files are already in ...\visual_studio_prj\Model1 directory! what could be the problem?

3. what i wanted to do actually was to test if i can completely put mini program aside when i'm developing the code. now that i couldn't do that, may i ask if this is possible? does the changes i make in the VS project actually return to the mini program (i highly doubt!), if not, then should i sync the two manually?

Thanks very much

freshhope, 2023-06-25:
1. compiler different, add all returns in your mp custom functions, return value should be s32 in your custom function
2. manually add c file in your vs project, TSMaster will do that automatically but vs don't
3. if you switch to vs code, you will never be able to continue your work in TSMaster

freshhope, 2023-06-27:
you can add c file here, do not use directory
[image]

arohami, 2023-06-27:
I think i figured it out!
it's both weird and kind of stupid!
after adding the c generated file (from simulink, or in my previous example the test.c), I changed the .c to .cpp. now the header is linked against the source file without any problem and compiler doesn't complain any more!

arohami, 2023-06-27:
now that i have finally managed to go to the next step, I faced another problem :(
when i run the code on visual studio in debug mode (similar to the video), none of my break points are hit even when i'm sending messages on CAN. 
to understand the problem better, i re-run the code normally (not debug mode), and i saw the TSMaster window (starting page) pop up. i checked the terminal and it seems the code is actually running TSMaster App!
I feel i'm still doing something wrong! if it helps and needed, i can specify the steps with more detail.  

[image]

freshhope, 2023-06-27:
please use attach method for debugging, do not direct start TSMaster
you can google how to attach to a process, or follow this video, sorry for the language, I do not have time to make english video, may be later
[image]

alternately, you can generate a VC++ project by TSMaster, TSMaster will auto set attach mode in the generated vs project, then you get to know how to set attach mode

arohami, 2023-06-27:
I don't know how to attach TSMaster to VC++. but i followed the same video you mentioned
That TSMaster windows in the image i sent is actually was opened when i run the the code in VC++ project. so i guess it should somehow connect. 
I also googled and couldn't find much about that. 
that could probably help if i had access to tutorial in text format (at least i can use google translate on that) or the video in english. Hope that happens soon.

Thanks again

freshhope, 2023-06-27:
steps:
[1] open a c code editor, write some code, then generate vs project to a directory, do not close TSMaster process
[image]
[image]

[2] open this vs project in visual studio, open a cpp file, set breakpoint in your code:
[image]

[3] just launch this vs project by F5, you may see that your breakpoint is not valid(with a yellow warning icon), don't warry, that is because your dll has not been loaded into TSMaster yet
[image]

[4] now switch to TSMaster, click this menu button "run last compiled", do not click the green run button !!! do not compile in TSMaster again !!!
[image]

[5] you will see your code is immediately paused by visual studio, and the breakpoint is valid now, because your dll is executed by TSMaster
[image]

[6] now you can use visual studio's powerful debugging feature in your own code!

arohami, 2023-06-28:
Thanks really. I think i misunderstood the expected behavior at first!
1. I thought other than debugging, i can run mini program code directly from Visual Studio (that results in opening a new TSMaster window). But going in the debug mode seems to be working without any problem. 

2. I had no idea how Visual Studio was connecting to TSMaster, so i was trying to get the same result without playing the compiled code in TSMaster (and only from VC++). however i checked configurations on vc++ and realized that it's actually sharing the same directory for resultant binary file with TSMaster and that's exactly why we are using "run last compiled" in TSMaster to avoid re-writing those vc++ generated files. 

generally, i guess it all comes to my lack of knowledge about visual studio features and debugging process!
i still don't fully understand how attaching visual studio debugger to some live process works, but thing's are working and that's good enough for me for now :)

closed 2023-06-28 by arohami

freshhope, 2023-06-28:
You are welcome! Attaching is a common debug method for all kinds of applications, not only in Visual Stdio, but also in Windbg and many other tools. Hope you are happy playing with TSMaster :)

arohami, 2023-06-29:
> You are welcome! Attaching is a common debug method for all kinds of applications, not only in Visual Stdio, but also in Windbg and many other tools. Hope you are happy playing with TSMaster :)

oh! then think i have very limited knowledge and experience with debugging applications!
I'll come around for sure because playing with apps always raises more questions :)