TOSUN-Shanghai/TSMaster #112: [libtscan_sdk_cpp_api] API is not complete

saerpa, 2023-09-05:
Hi, developers:

I'm trying to use TOSUN device (TC1014) in my Qt project. 
I download [libtscan_sdk_cpp_api.zip](https://www.tosunai.com/wp-content/uploads/2023/06/libtscan_sdk_cpp_api.zip) and I found that the API `tsfifo_add_can_canfd_pass_filter` is documented but not show up in `TSCANLINApi.h`.

I check with several other APIs and it works fine. Now I really need this missing API.
I try to add this one like others, but it doesn't work, it seems not in the .dll file.

ColinGong123, 2023-11-17:
Hi, thank you for your feedback.
Maybe you can update the latest LibTSCAN api which include the function. 

[https://github.com/TOSUN-Shanghai/libTSCANDemos/tree/main](url)

saerpa, 2023-12-11:
Okay, good job :)

closed 2023-12-11 by saerpa