TOSUN-Shanghai/TSMaster #114: 当PCAN USB Pro FD配置成CAN FD模式的时候，即使发送的数据类型是标准数据帧，实际发送的也是FD数据帧

lywind, 2023-10-03:
如图所示，我在发送类型中选择的是标准数据帧，但实际发送的是CAN FD数据帧。
[image]

TSSkyhits, 2023-11-16:
TSMaster软件版本是？

ColinGong123, 2023-11-17:
1. 不知咱们使用的TSMaster软件版本是多少呢？
2. 另外，最近由于部分客户在适配PEAK卡过程中遇到问题，新版本的TSMaster中PCAN FD 功能暂时禁用，今后将仅在TSMaster年度订购版本中提供支持。请知悉。

lywind, 2023-11-23:
> 1. 不知咱们使用的TSMaster软件版本是多少呢？
> 2. 另外，最近由于部分客户在适配PEAK卡过程中遇到问题，新版本的TSMaster中PCAN FD 功能暂时禁用，今后将仅在TSMaster年度订购版本中提供支持。请知悉。

TSMaster版本是V2023.9.28.975

ColinGong123, 2023-11-23:
> > 1. 不知咱们使用的TSMaster软件版本是多少呢？
> > 2. 另外，最近由于部分客户在适配PEAK卡过程中遇到问题，新版本的TSMaster中PCAN FD 功能暂时禁用，今后将仅在TSMaster年度订购版本中提供支持。请知悉。
> 
> TSMaster版本是V2023.9.28.975

嗯，咱们这个版本应该没有支持PEAK CAN FD的功能了。
关于其他厂商的CAN卡支持，我们的计划是将在年度订购版本中提供支持，可留意我们后续的发布，感谢关注。