TOSUN-Shanghai/TSMaster #115: PCAN Online Replay not working again in v2023.10.8.982

big-jar, 2023-10-10:
Hello,

After upgrade to v2023.10.8.982 the PCAN-USB pro can neither capture any FD frame nor online replay.  See figure 1

But roll back to v2023.5.7.864 it can work, See figure 2

figure1
[image]

figure2
[image]

freshhope, 2023-10-10:
We apologize for the inconvenience caused. Due to issues encountered by some customers during adaptation, the PCAN FD feature is temporarily disabled and will only be supported in the annual subscription version in the future.

lywind, 2023-10-10:
> We apologize for the inconvenience caused. Due to issues encountered by some customers during adaptation, the PCAN FD feature is temporarily disabled and will only be supported in the annual subscription version in the future.

What is the anual subscription version? Will it be paid?