TOSUN-Shanghai/TSMaster #120: PCAN-Ethernet Gateway Support

felixFleetZero, 2023-12-13:
Hi,
We've successfully used TSMaster with our Peak System PCAN-USB devices.
However, we can't seem to see PCAN-Ethernet Gateway device in the channel selection menu.
Note that we successfully see it on the same computer in Peak PCAN-View software.
Is it possible to interface that device with TSMaster?
[image]
[image]

ColinGong123, 2024-07-05:
We do not support Ethernet devices from other manufacturers yet

felixFleetZero, 2024-07-05:
Is this feature planned in the future? If so, what would be the timeline?

ColinGong123, 2024-07-08:
Sorry, we will not have any plans to support this device in the future