TOSUN-Shanghai/TSMaster #131: Does TSMaster Support CAN FD for Kvaser Leaf V3?

CL-Chiou, 2024-12-03:
I am using a Kvaser Leaf V3, which supports both CAN and CAN FD. However, when using TSMaster, I can only enable CAN, and there seems to be no option to configure CAN FD. On the other hand, when I use Kvaser CANking 7.0.1, I can successfully find and enable CAN FD for transmission and reception.

I would like to know if TSMaster currently supports enabling CAN FD on the Kvaser Leaf V3. Additionally, I noticed another issue: this behavior only occurs on the x86 version of TSMaster. When I run the x64 version, the Kvaser Leaf V3 device is not detected at all.

Could you please confirm whether TSMaster supports these features or if there are any configurations I might have missed?

Thank you!

ColinGong123, 2024-12-17:
Hello Chiou，

Thank you for your feedback,  Yes, TSMaster can support the CAN FD function with Kvaser Leaf V3, it require a license to support.

you could find our license client below the link and the "Vendor Interface Kvaser" for support.
also you can try to get a Free Trial license to test.
[https://license.tsmaster.net/mainInterface/pricing](url)

frankie-zeng, 2025-03-12:
@CL-Chiou [EcuBus-Pro](https://github.com/ecubus/EcuBus-Pro) support Kvaser CAN-FD.